openhardwaremonitor/openhardwaremonitor #1417: No CPU Temp (and possibly more) shown for AMD 5000 Series.

Ndrinta, 2021-04-01:
I know it takes a while for you guys to implement every new model and debug it but is there any roadmap yet?
When can we expect it to be working?

Cheers

Kyle2142, 2021-04-11:
This might be related but on my ryzen 3700 the CPU temperature is gone since I enabled Virtualisation (SVM I think) in bios. Wondering if the program can no longer access the info (because of virtualisation) or it must use a different API which it does not read from

Ndrinta, 2021-04-11:
Solved with Release Version 0.9.6
My bad, haven't noticed the update and I was still running 0.9.5

closed 2021-04-11 by Ndrinta

Kyle2142, 2021-04-11:
Just to close the loop on this, my bad - my *wattage* is missing since enabling SVM. Will search for or open a new issue

citelao, 2024-01-08:
I came across this issue while searching about my Ryzen 5800X not appearing in OHW (it was before). I indeed think it's because I have virtualization (Hyper-V) turned on: #1261