openhardwaremonitor/openhardwaremonitor #1419: High and increasing memory > 3,5 GB on my machine

47k, 2021-04-02:
Hey guys 

I'm using OpenHardwareMonitor on my Windows machines with the CheckMK Monitoring. On ony Windows 10 Client machine (all others are Microsoft Servers) the memory of the running program is increasing over 3,5 GB. 

Processes: 1
virtual: 8.11 GiB (warn/crit at 5.00 GiB/6.00 GiB)CRIT
physical: 3.50 GiB (warn/crit at 500 MiB/1000 MiB)CRIT
CPU: 0.02%
OpenHardwareMonitor.exe with PID 5888 CPU: 0.04%
Process handles: 385
Running for: 4 days 11 hours

[image]

m-j-carter, 2021-04-04:
I have been having the same issue. Are you running any additional plugins? I'm using [this Rainmeter plugin](https://github.com/abichinger/Rainmeter-HardwareMonitor) which is the likely culprit if the memory leak isn't in OHM itself, but it seems odd that you would be having the same issue for a completely unrelated reason. 

This is the memory usage after an uptime of around 48 hours:
[image]

version 0.9.6
Win10 64 bit, Ryzen 5 3600, 32GB 3200MHz RAM, RTX 3070.

47k, 2021-04-04:
Hey m-j-carter
No I have only the the OpenHardwareMonitor as a service running over NSSM (to mount it as a service). I see all the information in my CheckMK.

wkrueger, 2021-05-05:
Happens here too, I have the chart open.

mistohise, 2021-05-14:
Same issue here, v 0.9.6 open for a few days. 
[image]
Left on the sensor page showing CPU load/temps

doctorpangloss, 2021-07-08:
I also have this issue, showing 1.5GB of usage leaving OpenHardwareMonitor running for a while.

eschota, 2021-07-24:
++ for me

maty974, 2021-08-12:
Same same :(