openhardwaremonitor/openhardwaremonitor #1430: No CPU Temp for i3 10105F

pdp8989, 2021-04-29:
No Temperature details are viewable for an Intel i3 10105F.  Will the next update fix this issue?
[image]

Cereal-Killa, 2021-05-05:
Send me your cpu report so I can have a look at it (File -> Save Report)

pdp8989, 2021-05-05:
The CPU temperature seems to have fixed itself up.  However it is missing the motherboard temperature and CPU/SYS Fan speed
[image]
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/6430516/OpenHardwareMonitor.Report.txt)

Cereal-Killa, 2021-05-06:
Let me know if you can run this debug build and post the report here if it still doesn't work.

[Debug.zip](https://github.com/openhardwaremonitor/openhardwaremonitor/files/6434712/Debug.zip)

Cereal-Killa, 2021-05-06:
Seems like LibreHardwareMonitor already has support for that sensor, its forked from this project and has been updated more frequently, give it a try.

The releases page is outdated but you can download the latest revision from the artifacts:

Go here:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/809652028

or grab any successful build from here:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions

Scroll down to artifacts and download "LibreHardwareMonitor-net452".

pdp8989, 2021-05-06:
Debug build: 
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/6437958/OpenHardwareMonitor.Report.txt)
OpenHardwareMonitor.Report