openhardwaremonitor/openhardwaremonitor #1439: 11 Gen temperature Support

alexanderLog, 2021-05-16:
Does the application support temperature data for Intel gen 11? I have 11400f and it does not show any temps

mattbruman, 2021-05-31:
I second this request

KyleNie, 2021-06-29:
It works on my 11th gen laptop i9. Maybe it's a desktop specific issue?

MoritzMix, 2021-07-22:
Not working on 11900K

kneave, 2021-07-23:
Intel i9 11900F for me and it doesn't work

m-a-r-m-i-c, 2021-07-30:
i7-11700F here, doesn't work for me too.

Would appreciate an update ;)

BrsVgl, 2021-08-14:
Hallo everyone,

I have added Intel 11th Gen CPU support (Rocket Lake) and also Intel Jasper Lake.
I have made a pull request and you can also download this as a pre-release from my fork:
[BrsVgl/openhardwaremonitor/0.9.6.1-PR](https://github.com/BrsVgl/openhardwaremonitor/releases/tag/0.9.6.1-PR)

Any tests from your side (with screenshot as a proof) are highly appreciated.

FedyuninV, 2021-08-14:
> Hallo everyone,
>
> I have added Intel 11th Gen CPU support (Rocket Lake) and also Intel Jasper Lake.
> I have made a pull request and you can also download this as a pre-release from my fork: [...]

i7-11700k confirmed.
[image]

kneave, 2021-08-14:
Thank you! 11900F confirmed, hopefully the Z590 will be added before long too. Any pointers on how to do this? Happy to have a look but no idea where to start.

[image]

m-a-r-m-i-c, 2021-08-15:
I'm able to confirm that I7-11700F is working.
Thanks for your efforts.

BrsVgl, 2021-08-15:
> Thank you! 11900F confirmed, hopefully the Z590 will be added before long too. Any pointers on how to do this? Happy to have a look but no idea where to start.
> 
> [image]

I can have a look at it. I suspect this is the same as with Rocket Lake: The sensors and everything are basically like the predecessor. We just need to let the library know how to detect the Z590.

kneave, 2021-08-15:
Awesome, thank you!

[image]<https://user-images.githubusercontent.com/733411/129461744-13dc3dc0-4475-47c6-b3c9-0887677eb48f.png>

mattbruman, 2021-08-17:
Working on my 11700K as well. 

[image]

jreed2535, 2021-10-03:
Confirmed pre-release 0.9.6.1 resolved issue/missing data for ASUS ROG STRIX Z590-E GAMING WIFI board and 11th Gen Intel Core i7-11700k.  

0.9.6.1-PR screenshot
[image]

Tyux, 2021-12-07:
> Hallo everyone,
>
> I have added Intel 11th Gen CPU support (Rocket Lake) and also Intel Jasper Lake. I have made a pull request and you can also download this as a pre-release from my fork: [...]

Not work for my 11700K
[image]