openhardwaremonitor/openhardwaremonitor #1440: M.2 Drive isn't shown properly

nvypwr, 2021-05-29:
Hi, I've recently added a M.2 drive to my PC. But OpenHardwareMonitor doesn't show it properly. 
[image]

As seen in the screenshot, it's shown as "Generic Hard Drive" plus, it's temperature sensor is missing as well. Please fix this issue.

kristx300, 2021-06-07:
I think solution placed here [CrystalDiskInfo](https://github.com/hiyohiyo/CrystalDiskInfo/blob/eb348018f2abc8eaf634271e665d4c379829fe48/AtaSmart.cpp#L7251)
With different releases for other marks like [Samsung](https://github.com/hiyohiyo/CrystalDiskInfo/blob/eb348018f2abc8eaf634271e665d4c379829fe48/AtaSmart.cpp#L6678)

pgrawehr, 2021-06-14:
You can try whether my fork [here](https://github.com/hexagon-oss/openhardwaremonitor/releases/tag/v0.9.7-alpha01) fixes this - at least it works for me.

cagnulein, 2021-09-01:
@pgrawehr unfortunately doesn't fix the issue for me :( the Generic Hard disk label disappear at all with your fork. Any data that could be useful for fix the issue?