openhardwaremonitor/openhardwaremonitor #1441: Request for winget integration

Pingger, 2021-06-04:
Hi,

would you see if you can add openhardwaremontior to the winget?
https://github.com/microsoft/winget-cli
https://github.com/microsoft/winget-pkgs

Greetings

PowerRangerX, 2023-08-05:
Is this being considered? Cause it'll be very usefull, plus other monitoring softwares are also on `winget` like [HWMonitor](https://www.cpuid.com/softwares/hwmonitor.html) and others.

JaneX8, 2024-04-18:
I would also really like to see this on Winget to include it in automation.

nlxdodge, 2024-11-20:
They don't have any releases on Github. Then I would be able to copy and paste a script to run and automatically create a pull request for the Winget Packages Repository. 

I can however, manually insert the program once. And if it has an auto updater, that will solve that problem then.

JaneX8, 2024-11-22:
This works.

```ps1
scoop bucket add extras
scoop install extras/openhardwaremonitor
```