openhardwaremonitor/openhardwaremonitor #1442: What is the sampling time for measurements (and is it changeable?)

absolution1, 2021-06-07:
Hi everyone,

As the title says, what is the sampling time for any of the measurements that OHW reports?  
Just for clarity's sake, every time OHW reports a CPU core temperature, over what time period was that temperature measured?  Is it 0.1s, 1s, 5s etc?

Secondly, is it possible to tweak/change that sampling time?

pgrawehr, 2021-06-14:
@absolution1 The sample time is 1 second for most sensors. Some (like the HDD sensors) have it set to 30 seconds. This cannot currently be changed. The temperature measurements are _current_ values, not a measurement over a period.

absolution1, 2021-06-14:
@pgrawehr thanks for letting me know.
I know that with windows performance counters you can request CPU load over a time period e.g. load over the last 1 minute, 10 minutes etc.  I was wondering if it was possible to get some kind of equivalent temperature measurement (like an average) over a time period with OHW. 
It sounds like I'd need to have OHW log to a file and have a script periodically scrape the data to calculate an average, or something similar...

Anyway, thank you for the info!

pgrawehr, 2021-06-14:
@absolution1 Yes, there's no rolling average feature right now.