openhardwaremonitor/openhardwaremonitor #1451: Any way to save data periodically to avoid losing data if PC crashes?

johnverv, 2021-07-10:
Hi guys, sorry this may have been asked but I didnt find any similar questions;
I use OHM heavily for overclock analysis, but I find it strange that if my PC crashes I lose all the recorded data from that session in OHM, the only way around this for me is to periodically close and open OHM manually. I suspect either there is a way to have it commit data to disk, or a technical reason why this doesnt happen automatically, just thought I would ask the question!

For example I recently tweaked some voltage settings and let my system run a stress test for over an hour without issue, however during a gaming session where the system was nowhere close to stress test temp \ voltages, my system switched off.
Rebooting and loading OHM under these conditions would open up with a blank data set of the entire gaming session and the stress test, as I had not rebooted or closed OHM since starting the stress test.

Hopefully this explains what I am rambling on about

Thanks!

shrddr, 2021-07-11:
Tick Options > Log Sensors to save the values to a csv file periodically

marioscience, 2023-07-23:
This works nicely when the PC is functioning correctly. The file seems to not be committed to disk periodically, but only on OHM closing successfully. When there's an unexpected crash, all data will be gone, almost as if the file itself is wiped out. 

I bought a GPU and I wanted to check the temps, turns out the GTX1080ti I bought was defective because when playing a demanding game after a while the system would crash. I wanted to see if it was an issue with temperature going over the spec'd temperature for the GPU but never could because of this issue. 

It would be really nice if the application would actually close and open the file it is logging the sensors to periodically to at least recover some of the data when there's a system crash. 

The app is great and works wonders, but this is somewhat of a downside for me. It is nice to see the data when everything works as expected, but it would be specially useful if I could go back and see some of the data when a crash happens.

marioscience, 2023-07-23:
I dug a bit into the code, C# is not my strong suit. The file is opened in AppendMode when  it's being written to, so it should work as expected. So this seems to be an issue with StreamWriter object itself. Maybe a solution would be to add an option to recycle opening and closing the file with a specified time period. Say 5 seconds, 10 seconds, 1 minute, etc. and then force C# to close the file and reopen it. For performance, this could be done (somehow) while the visiting methods in Hardware.cs are collecting 4 of the data points. 

Ugh, that's a case of the software being well written but the underlying dependencies dropping the ball... If I was more confident with C# I would tinker with the code and offer a pull request.. but I will probably blow it. Hopefully, the devs add this. It would be an awesome feature to diagnose hardware issues.