openhardwaremonitor/openhardwaremonitor #1453: HDD hard disk temperature indicator vanished due to recent .Net update

MikhailRokhin, 2021-08-07:
After recent update

2021-07 Preview version update .NET Framework 3.5 and 4.8  Windows 10 Version 21H1 x64 (KB5004331)

HDD temperature indicator became non-available and HDD changed from detected WDC to "Generic hard disk" .

Windows version is 10.0.19043.1110

Open HW Monitor version is the latest 0.9.6

[image]

cagnulein, 2021-09-01:
did you fix it somehow?

MikhailRokhin, 2021-09-02:
> did you fix it somehow?

No, just waiting for developer... ))

cagnulein, 2021-09-03:
@MikhailRokhin check this https://github.com/hexagon-oss/openhardwaremonitor/issues/15

MikhailRokhin, 2021-09-03:
> @MikhailRokhin check this [hexagon-oss#15](https://github.com/hexagon-oss/openhardwaremonitor/issues/15)

Thank you!

By the way, do you have some strange behaviour of OHWM after wake-up PC from sleep/hibernation : it halts and takes few minutes to refresh the output of widget?

In general, I think it was bad idea to get into with .Net as it is deadly faulty.

cagnulein, 2021-09-04:
mmm i didn't check @MikhailRokhin i'm just using OHWM to track data when i have strange conditions...