openhardwaremonitor/openhardwaremonitor #1461: Support for Gigabyte Z590 GAMING X. Super IO chip ITE IT8689E

mana8100, 2021-08-24:
Support for Gigabyte Z590 GAMING X. Super IO chip ITE IT8689E

I actualy added it at the source copy i got and works as a charm so far. I just had to add the mother model and the chip with the same functinality as the IT8688E. And works like a charm. Not sure tho if i am missing anything.

Gigabyte Z590 GAMING X
chip ID 34441 (0x8689) -> ITE IT8689E

https://gyazo.com/ade4c8876deb64808b2f415e8c1999dc

kneave, 2021-09-08:
Could you share where you made these changes please? I'm trying to get it working myself and I'm not sure where to look.

kneave, 2021-09-08:
Ah, got it! I'll submit a PR shortly :)

kneave, 2021-09-08:
With the chip ID you provided I've managed to get it working with the Z590I Aurus Ultra, I don't know how to make it work for all Z590 boards though.

My fork is here if anyone wants to have a look: https://github.com/kneave/openhardwaremonitor