openhardwaremonitor/openhardwaremonitor #1472: Cannot find how to uninstall

Plex-newbie, 2021-10-30:
I wanted to uninstall the program, but the program is not listed in "Apps and Features" for Windows 10.  Please advise how to safely remove the program.

batmov, 2021-11-01:
The program is a standalone application. Just delete the executable.

sushil-2803, 2022-01-07:
You can just delete the folder contaning the files which you downloaded. There is no need for uninstaller.

mi1stormilst, 2025-08-08:
There are several post where people have not been able to delete the folder. There is something still running and I have checked all tasks and services and rebooted.

PCAssistSoftware, 2025-08-08:
I suspect it is open in your system tray e.g. near clock.

1) click on arrow to show all apps running in system tray
2) right click on Open Hardware Monitor and choose exit
3) then you can delete folder

[image]

mi1stormilst, 2025-08-08:
That was the first thing I did. Weirdly I downloaded the files again and I extracted them into the same folder and I was able to delete it. No idea!