openhardwaremonitor/openhardwaremonitor #1474: Intel Core i7 12700 Incomplete info

asdf130162, 2021-11-11:
[image]
When I debug code，The received data type is only load

asdf130162, 2021-11-13:
Updates from my Fork  [https://github.com/asdf130162/openhardwaremonitor](https://github.com/asdf130162/openhardwaremonitor)

electricJP, 2021-11-30:
Thank you very much for this

CHarlit0n, 2021-12-10:
Could you help me fix the same isseu?

mkanet, 2023-06-16:
@asdf130162  Do you know of a version that supports _current_ Intel CPUs such as the I7 13700K?  I can't seem to find a single Windows tool that provides CPU core temps in realtime via command-line for the I7 13700K.

SiavaRu, 2023-11-03:
> Updates from my Fork https://github.com/asdf130162/openhardwaremonitor [image]

CPU total % on your screenshot is incorrect. Do you know how fix it?