openhardwaremonitor/openhardwaremonitor #1480: Mitac Motherboard PH10CMU with Nuvoton NCT6126D don't work

mhilfiker, 2021-12-08:
Hi,

It seems that the sensor NCT6126D is not known, there isn't an entry for the Mainboard sensors.

Manual: https://download.mitacmct.com/Files/manual/CBU/Motherboard/PH10CMU%20Product%20Guide%20v1.1.pdf
API Software:  https://www.mitacmct.com/IndustrialMotherboard_PH10CMU_PH10CMU

Is it possible to integrate the NCT6126D in future versions?

Thanks for your hard work on this Software
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/7676489/OpenHardwareMonitor.Report.txt)
!

DarthDestroyer, 2021-12-08:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD283 at 0x2E/0x2F

I have seen that ID before! Did you run a search in the issues?

mhilfiker, 2021-12-10:
I ran now the search and find with "winbond" some entries with the same IDs, but no solution for this. Did you know, is the Problem a Driver which is not avaiable or is it just unknown by ohm?

Thanks

DarthDestroyer, 2021-12-10:
It will not be a easy Solution!

But you can check #1101 for some Infos and how he got it to work!

The thing is, its pretty hard to help from far... I dont use your Mobo, so it would be awesome if you try it and contribute to us!

I would like to help, if you need me