openhardwaremonitor/openhardwaremonitor #1481: No Temperature Sensor for Intel 11th GEN 11700 CPU

cudeech, 2021-12-10:
No temperature sensor for the 11th gen intel 11700 CPU

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/7694633/OpenHardwareMonitor.Report.txt)

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.9.6.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.22000.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Dell 0K3CM7 Motherboard (/mainboard)
|
+- Intel 11th Gen Core i7-11700 CPU (/intelcpu/0)
|  +- CPU Total Usage : 0.585938        0  39.1602 (/intelcpu/0/load/0)
|  +- CPU Core #1    :   1.5625        0       50 (/intelcpu/0/load/1)
|  +- CPU Core #2    :   1.5625        0  51.5625 (/intelcpu/0/load/2)
|  +- CPU Core #3    :  0.78125        0  47.6191 (/intelcpu/0/load/3)
|  +- CPU Core #4    :        0        0  47.6191 (/intelcpu/0/load/4)
|  +- CPU Core #5    :        0        0  46.8254 (/intelcpu/0/load/5)
|  +- CPU Core #6    :        0        0  42.1635 (/intelcpu/0/load/6)
|  +- CPU Core #7    :        0        0  43.7748 (/intelcpu/0/load/7)
|  +- CPU Core #8    :  0.78125        0  47.6191 (/intelcpu/0/load/8)
|
+- Dell System Memory (/ram)
|  +- Memory Usage - 32GB :  18.4203  12.3253   22.383 (/ram/load/0)
|  +- Used Memory    :  5.82277   3.8961  7.07543 (/ram/data/0)
|  +- Available Memory :  25.7879  24.5353  27.7146 (/ram/data/1)
|
+- NVIDIA GeForce RTX 3060 Ti GPU (/nvidiagpu/0)
|  +- GPU Core       :      210      210     1665 (/nvidiagpu/0/clock/0)
|  +- GPU Memory     :      405      405  7000.98 (/nvidiagpu/0/clock/1)
|  +- GPU Shader     :        0        0        0 (/nvidiagpu/0/clock/2)
|  +- Temperature    :       52       33       54 (/nvidiagpu/0/temperature/0)
|  +- GPU Core       :        0        0       99 (/nvidiagpu/0/load/0)
|  +- GPU Frame Buffer :       36        0       80 (/nvidiagpu/0/load/1)
|  +- GPU Video Engine :        0        0       48 (/nvidiagpu/0/load/2)
|  +- GPU Bus Interface :        0        0        7 (/nvidiagpu/0/load/3)
|  +- GPU Memory Usage - 8GB :  16.7893  8.21066  32.1592 (/nvidiagpu/0/load/4)
|  +- GPU            :        0        0     1102 (/nvidiagpu/0/fan/0)
|  +- GPU Fan        :        0        0       30 (/nvidiagpu/0/control/0)
|  +- GPU Power      :   11.622    9.477   47.726 (/nvidiagpu/0/power/0)
|  +- GPU Memory Free :  6816.62  5557.52  7519.38 (/nvidiagpu/0/smalldata/1)
|  +- GPU Memory Used :  1375.38  672.617  2634.48 (/nvidiagpu/0/smalldata/2)
|  +- GPU Memory Total :     8192     8192     8192 (/nvidiagpu/0/smalldata/3)
|  +- GPU PCIE Rx    :        0        0  651.367 (/nvidiagpu/0/throughput/0)
|  +- GPU PCIE Tx    :        0        0  70.3125 (/nvidiagpu/0/throughput/1)
|
+- Seagate BarraCuda 7200 RPM HDD (/hdd/0)
|  +- Temperature    :       31       23       32 (/hdd/0/temperature/0)
|  +- Used Space - 1TB :  26.3299  26.3299  26.3299 (/hdd/0/load/0)
|
+- Western Digital NVMe SSD (/hdd/1)
|  +- Temperature    :       47       37       47 (/hdd/1/temperature/0)
|  +- Used Space - 1TB :  58.4608  58.4327   58.463 (/hdd/1/load/0)
|
+- Verbatim USB HDD  (/hdd/2)
|  +- Used Space - 1TB : 0.0232163 0.0232163 0.0232163 (/hdd/2/load/0)
|
+- Toshiba USB SSD (/hdd/3)
[... 1302 lines cut ...]
--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Generic Hard Disk
Firmware version: Unknown

Logical drive name: E:\
Format: NTFS
Total size: 1000186310656
Total free space: 1000048300032

--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Generic Hard Disk
Firmware version: Unknown

--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Generic Hard Disk
Firmware version: Unknown

FreePaul, 2022-01-06:
Yes! Very need Intel 11th GEN support!!!

briankelly, 2022-01-06:
If you're open to compiling your own, the branch on PR #1431 (from @Cereal-Killa) worked for me.

Cereal-Killa, 2022-01-06:
Just merged upstream with my repo and built a release: https://github.com/Cereal-Killa/openhardwaremonitor/releases/tag/0.10.0-alpha1

derailius, 2022-09-19:
> Just merged upstream with my repo and built a release: https://github.com/Cereal-Killa/openhardwaremonitor/releases/tag/0.10.0-alpha1

you're awesome, thanks!

mkanet, 2023-06-16:
Hi, I'm not sure if anyone is still watching this open issue.  When I try to execute version 0.10.0-alpha1, it tells me I need to install .NET runtime; and, attempt to download a super-old .NET Runtime 5.0 version; which is 2 major versions behind.

@Cereal-Killa or @derailius **Is .NET Runtime 7.0 compatible with this alpha release?**  Will it work right out of the box?  Or, is there something I need to do after installing .NET Runtime 7.0?