openhardwaremonitor/openhardwaremonitor #1490: Can't obtain CPU voltage and frequency

CodeZ1LLa, 2021-12-29:
I use a for loop to check if the sensor contains names in order to pick up needed sensors.  
In short, it looks like this: 
```
for (int i = 0; i < computer.Hardware.Length; i++)
            {
                IHardware hardware = computer.Hardware[i];
                if (hardware.HardwareType == HardwareType.CPU)
                {          
                    hardware.Update();

                    for (int i1 = 0; i1 < hardware.Sensors.Length; i1++)
                    {
                        ISensor sensor = hardware.Sensors[i1];
                        switch (sensor.SensorType)
                        {
                            case SensorType.Temperature when sensor.Name.Contains("CPU Package"):
                                CpuTemp = sensor.Value.GetValueOrDefault();
                                break;   

                            case SensorType.Voltage when sensor.Name.Contains("CPU Voltage"):
                                VCore = sensor.Value.GetValueOrDefault();
                                break;
```

And while I easily can read the CPU package, regardless of what names I tried to look for (CPU Voltage, VCore, CPU Vcore, etc.) it doesn't find the voltage and frequency. Not sure if I looking for the wrong name or lib just doesn't support Ryzen 5000. 

Using OHM Lib NuGet 0.9.6
CPU: Ryzen 5900X
Language: C#
Latest Windows 10 updates
BIOS: 4603 with latest AGESA

What could be a problem? The name?

DarthDestroyer, 2021-12-30:
Its easy to check, if your hardware is supported!
Just run OHM by itself.

For the names, just print them (for example the sensor entity) to the debug console.
It should be an object, depending on your coding software it can export an object to console or you would need a short function for object to string before writing to console...

that would be the easiest way

CodeZ1LLa, 2022-01-01:
Thanks! I tried to print them but I can't find the name that would correspond with a CPU voltage. My CPU is supported, and I can invoke the CPU Package Power, but not the voltage or frequency. I wonder if there is a list of names somewhere, that can be used.

DarthDestroyer, 2022-01-01:
You can also look at the Report.txt File

Just hit File, Report and there should be a save Button