openhardwaremonitor/openhardwaremonitor #1498: Windos XP 32 bit

alexi3, 2022-01-30:
Hi,
on https://openhardwaremonitor.org/ is written: "Hardware Monitor software runs on 32-bit and 64-bit Microsoft Windows XP"
I am using Windows XP SP3 32 bit
OpenHardwareMonitor 0.7.1 and 0.8.0 is running.
When starting OpenHardwareMonitor.exe 0.9.6 I get "OpenHardwareMonitor.exe ist keine zulässige Win32 Anwendung"

alexi3, 2022-02-12:
Please have a look at Issue #1499
I have Microsoft Visual C# 2008 Express Edition.
I could successfully compile Open Hardware Monitor 0.1.19.0 (Feb. 12, 2010).
I can't compile current Open Hardware Monitor.
Open Hardware Monitor is good software and I want to help to improve it.

DarthDestroyer, 2022-02-12:
you know, that you can get the actual community edition for free? Just an Information...

What error does it tell you, if you try to compile the last Version?

which .net version do you use with XP? Or which is installed?

alexi3, 2022-02-12:
Yes I know but [https://visualstudio.microsoft.com/de/vs/community/]
shows 
Visual StudioSetup.exe is no vailid Win32-application (trnsalte into english by me).

.net:
[image]

errors:
1st compile:
[image]
2nd compile because of no success of 1st compile:
[image]

alexi3, 2022-02-16:
Same Problem when starting OpenHardwareMonitor.exe 0.9.6 "OpenHardwareMonitor.exe ist keine zulässige Win32 Anwendung" after installing .NET 4.

esiefker, 2024-09-15:
I can confirm this bug.  

It would be nice if it would be fixed, but otherwise removing XP from the list of supported systems and making a note on the webpage of the last version supported by XP would be helpful.  We probably don't actually need new features on hardware running XP.

esiefker, 2024-09-15:
BTW, where can I download old versions of OHWM?