openhardwaremonitor/openhardwaremonitor #1506: Is this repo still maintained?

oomek, 2022-04-02:
If not is there any fork that keeps this alive?

msawyer91, 2022-04-02:
It seems this repo has been aging gracefully, but there are active forks. The [hexagon-oss fork](https://github.com/hexagon-oss/openhardwaremonitor) seems to be the most active, and I noticed that one has support for NVMe drives and a lot more. I created my own [fork](https://github.com/msawyer91/openhardwaremonitor) because there was an issue I wanted to fix. Since I don't have access to this original repo or the hexagon-oss one, I forked the hexagon-oss repo, made my changes, then submitted a pull request. If hexagon-oss merges it, my changes will become a part of their master.

oomek, 2022-04-02:
Thanks. I'm asking because the grace period has ended unfortunately with the release of the latest Radeon drivers and current Windows 11 release preview build. I've lost all Radeon sensors and looking for the repo that's up to date with AMD gpu support, or a clue how to add/fix it myself.

msawyer91, 2022-04-03:
Unfortunately I can't help you with AMD since I don't have any AMD cards. I run all NVIDIA gear. Definitely take a look at the build I posted, or the latest available release from the hexagon-oss fork, as either of those are way more up-to-date than the main repo. The only difference between mine and hexagon-oss is that I tweaked the NVIDIA GPU memory reporting because in both this repo, and hexagon-oss, when you have multiple GPUs, the memory report is correct only for the first GPU. All additional GPUs only repeated the same memory stats as the first.

The hexagon-oss folks did a really nice job. Things like NVMe drives show much better reporting, so definitely check it out and see if it solves your issue.

Did AMD stop making sensor data available with their latest drivers? I'd be really surprised that they would hide that info.

oomek, 2022-04-03:
I've found a website that list all forks in the order of activity, checked all that were ahead, but none of them addressed the AMD issue unfortunately. Tried to analyze the source. It's ADL api not returning the driver version properly and then spewing a `ERR_INVALID_ADL_IDX` error on function calls. Not sure where the problem lies really.

Here is the AMD part of the log:
```
AMD Display Library

ADL Status: OK
ADL2 Status: OK

Graphics Versions

 Status: OK_WARNING
 DriverVersion: 
 CatalystVersion: 
 CatalystWebLink: http://support.amd.com/drivers/xml/driver_09_us.xml

Number of adapters: 7

AdapterIndex: 0
isActive: 1
AdapterName: Radeon RX Vega
UDID: PCI_VEN_1002&DEV_687F&SUBSYS_0B361002&REV_C1_6&263BD31&0&00000008A
Present: 1
VendorID: 0x1002
BusNumber: 3
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x0

AdapterIndex: 1
isActive: 0
AdapterName: Radeon RX Vega
UDID: PCI_VEN_1002&DEV_687F&SUBSYS_0B361002&REV_C1_6&263BD31&0&00000008&02A
Present: 1
VendorID: 0x1002
BusNumber: 3
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x0

AdapterIndex: 2
isActive: 0
AdapterName: Radeon RX Vega
UDID: PCI_VEN_1002&DEV_687F&SUBSYS_0B361002&REV_C1_6&263BD31&0&00000008&03A
Present: 1
VendorID: 0x1002
BusNumber: 3
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x0

AdapterIndex: 3
isActive: 0
AdapterName: Radeon RX Vega
UDID: PCI_VEN_1002&DEV_687F&SUBSYS_0B361002&REV_C1_6&263BD31&0&00000008&04A
Present: 1
VendorID: 0x1002
BusNumber: 3
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x0

AdapterIndex: 4
isActive: 0
AdapterName: Radeon RX Vega
UDID: PCI_VEN_1002&DEV_687F&SUBSYS_0B361002&REV_C1_6&263BD31&0&00000008&05A
Present: 1
VendorID: 0x1002
BusNumber: 3
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x0

AdapterIndex: 5
isActive: 0
AdapterName: Radeon RX Vega
UDID: PCI_VEN_1002&DEV_687F&SUBSYS_0B361002&REV_C1_6&263BD31&0&00000008&06A
Present: 1
VendorID: 0x1002
BusNumber: 3
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x0

AdapterIndex: 6
isActive: 0
AdapterName: Radeon RX Vega
UDID: PCI_VEN_1002&DEV_687F&SUBSYS_0B361002&REV_C1_6&263BD31&0&00000008&07A
Present: 1
VendorID: 0x1002
BusNumber: 3
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x0

--------------------------------------------------------------------------------

AMD GPU

AdapterIndex: 0

Overdrive Caps

 Status: ERR_INVALID_ADL_IDX
 Supported: 0
 Enabled: 0
 Version: 0

Overdrive5 Parameters

 Status: ERR_INVALID_ADL_IDX
 NumberOfPerformanceLevels: 0
 ActivityReportingSupported: 0
 DiscretePerformanceLevels: 0
 EngineClock.Min: 0
 EngineClock.Max: 0
 EngineClock.Step: 0
 MemoryClock.Min: 0
 MemoryClock.Max: 0
 MemoryClock.Step: 0
 Vddc.Min: 0
 Vddc.Max: 0
 Vddc.Step: 0

Overdrive5 Temperature

 Status: ERR_INVALID_ADL_IDX
 Value: 0

Overdrive5 FanSpeed

 Status RPM: ERR_INVALID_ADL_IDX
 Value RPM: 0
 Status Percent: ERR_INVALID_ADL_IDX
 Value Percent: 0

Overdrive5 CurrentActivity

 Status: ERR_INVALID_ADL_IDX
 EngineClock: 0
 MemoryClock: 0
 Vddc: 0
 ActivityPercent: 0
 CurrentPerformanceLevel: 0
 CurrentBusSpeed: 0
 CurrentBusLanes: 0
 MaximumBusLanes: 0

Overdrive6 CurrentPower

 Power[TOTAL_POWER].Status: ERR_INVALID_ADL_IDX
 Power[PPT_POWER].Status: ERR_INVALID_ADL_IDX
 Power[SOCKET_POWER].Status: ERR_INVALID_ADL_IDX
 Power[CHIP_POWER].Status: ERR_INVALID_ADL_IDX

OverdriveN Temperature

 Temperature[CORE].Status: ERR_INVALID_ADL_IDX
 Temperature[MEMORY].Status: ERR_INVALID_ADL_IDX
 Temperature[VRM_CORE].Status: ERR_INVALID_ADL_IDX
 Temperature[VRM_MEMORY].Status: ERR_INVALID_ADL_IDX
 Temperature[LIQUID].Status: ERR_INVALID_ADL_IDX
 Temperature[PLX].Status: ERR_INVALID_ADL_IDX
 Temperature[HOTSPOT].Status: ERR_INVALID_ADL_IDX

OverdriveN Performance Status

 Status: ERR_INVALID_ADL_IDX
 CoreClock: 0
 MemoryClock: 0
 DCEFClock: 0
 GFXClock: 0
 UVDClock: 0
 VCEClock: 0
 GPUActivityPercent: 0
 CurrentCorePerformanceLevel: 0
 CurrentMemoryPerformanceLevel: 0
 CurrentDCEFPerformanceLevel: 0
 CurrentGFXPerformanceLevel: 0
 UVDPerformanceLevel: 0
 VCEPerformanceLevel: 0
 CurrentBusSpeed: 0
 CurrentBusLanes: 0
 MaximumBusLanes: 0
 VDDC: 0
 VDDCI: 0

Performance Metrics

 Status: ERR_NOT_SUPPORTED
```

mindplay-dk, 2023-06-16:
The `hexagon-oss` fork doesn't seem active anymore - here's a different fork that is currently active:

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

Perhaps this repository should be flagged as archived? And maybe consider adding a README to point folks to other active forks. I've searched a few times and they are not that easy to find.