openhardwaremonitor/openhardwaremonitor #1515: AMD Drivers 22.5.2 brokes OHM reading Radeon RX 580 sensors.

UnclWish, 2022-05-24:
After update AMD Radeon Drivers to 22.5.2 (Windows 11 Pro) OHM 0.9.6 stops to show clocks of GPU and vmem, voltages, etc.
Now it shows only temperature and power sensors.

isaaclepes, 2022-06-04:
Same issue for me after updating to 22.5.2 on Windows 10 Pro with RX 480.  OHM just shows the name of the GPU, but no sensors.  I can still see the temp in Task Manager.

[image]

isaaclepes, 2022-06-04:
I submitted my comment a bit too soon.. just realized I was on an older version of OHM.  Version 0.9.6 does show the sensors correctly for AMD RX 480.

[image]

UnclWish, 2022-06-05:
> I submitted my comment a bit too soon.. just realized I was on an older version of OHM. Version 0.9.6 does show the sensors correctly for AMD RX 480.
> 
> [image]

No, not correctly! Only Temperature and power. But no frequencies, load, fan etc...

Jlpvasconcelos, 2022-06-12:
Can someone fix this problem, or open hardware monitor is dead ???????

UnclWish, 2022-06-17:
Ohhh, it's good! Thanks!

UnclWish, 2022-06-17:
Sorry for offtop, but what need to add in OHM reading temperature sensors from DDR3 memory modules? Crucial Bullistix.

AndrewGG88, 2022-06-19:
> Hi, I made a fix to show the missing information on RX580 and 22.5.2 drivers
> 
> [image]
> 
> Visit my fork https://github.com/joaquimorg/openhardwaremonitor

Thank you very much!