openhardwaremonitor/openhardwaremonitor #1520: No CPU & MotherBoard temperatures (null value) with WPF application

washburnit, 2022-06-18:
Hi,
I'm succesfully using OpenHardwareMonitor DLL with a "standard" desktop app.
I "ported" this app to a WPF (Windows Presentation Foundation) desktop application but I noticed that all the temperature values (CPU and Mother Board) are always "null".
The DLL is the same of the other application, so should I do something different with OpenHardwareMonitor if used in a WPF app?
The .NET version is the same (4.7.2) and the program is compiled @ 32 bit.
Thank you, regards.

Roberto

rmattuschka, 2022-06-19:
You must use Tagetplatform Any CPU and no prefer to 32bit.
It's because the hardwaredriver

washburnit, 2022-06-19:
I discovered that it was because the program was not executed with administrator privileges, even though my user has administrator profile.
Then I gave administrator privilege to the program (through the app.manifest) and it was able to read data from temperature sensors otherwise it could read everything else but not those sensors.
Regards,

Roberto