openhardwaremonitor/openhardwaremonitor #1521: Serial port function

dod1k228, 2022-06-30:
[image]
[image]

hello.
Since 2018 I was using v.8.1.0 for outputting some pc info to arduino device
After video card upgrade old version was unable to display gpu info, had to dowload the latest version
and found that there is no more "Serial " option 

Kindly advise how this problem can be solved. Thank You very much in advance and sorry if same question was already brought up on the forum.

Krowssey, 2023-08-08:
@dod1k228 Did you find a solution?

dod1k228, 2023-08-12:
> @dod1k228 Did you find a solution?

Hello, yes I did

I switched to Libre Hardware Monitor (ver 0.8.9.0) instead of open hardware monitor (it is simular soft)

szczupakoffi, 2024-03-10:
can you get that as an app like open hardware monitor because i can't find it in this version

PCAssistSoftware, 2024-03-10:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

szczupakoffi, 2024-03-10:
> https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

yea but i need it as an app.like .exe

PCAssistSoftware, 2024-03-10:
Well download it from the releases page then and you will have the EXE....

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases/tag/v0.9.3

szczupakoffi, 2024-03-10:
> Well download it from the releases page then and you will have the EXE....
> 
> https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases/tag/v0.9.3

oh ok thanks i guess that was my brain issue but still no serial on that version

szczupakoffi, 2024-03-10:
> > @dod1k228 Did you find a solution?
> 
> Hello, yes I did
> 
> I switched to Libre Hardware Monitor (ver 0.8.9.0) instead of open hardware monitor (it is simular soft)

how did you get the serial on this version