openhardwaremonitor/openhardwaremonitor #1528: Cortex flagged software: Behavioral threat detected

Valkyrie19k, 2022-08-25:
[image]

Application information:
Application name:	Open Hardware Monitor
Application version:	0.9.6.0
Process ID:	23520
Application location:	C:\Users\*********\Downloads\OpenHardwareMonitor\OpenHardwareMonitor.exe
Command line:	"C:\Users\*********\Downloads\OpenHardwareMonitor\OpenHardwareMonitor.exe" 
File origin:	Hard drive on this computer
User name:	*********

Prevention information:
Prevention date:	Thursday, August 25, 2022
Prevention time:	10:05:20
OS version:	10.0.19044
Component:	Behavioral Threat Protection
Cortex XDR code:	c0400067
Prevention description:	Behavioral threat detected
Additional information 1:	Rule vulnerable_driver_dropped_WinRing0.sys

bandwith, 2022-09-16:
I'm having the same problem.

usamac, 2022-12-26:
Win 11 Build 22621.963 Win Defender 1.381.1117.0 is flagging OHM .9.6 without an option to override and allow. OHM is still working, but I wonder if it is only for a matter of time. 

[image]