openhardwaremonitor/openhardwaremonitor #1530: Cannot see NVMe SSD temperature

michalbalek, 2022-08-27:
I cannot see temperature for my two NVMe disk.
Disk types:
SAMSUNG MZVKW512HMJP-000H1
WDC PC SN730 SDBPNTY-1T00-1101

Checking via CrystalDiskInfo show temperature fine.

[image]

411621, 2022-09-22:
Confirmed: Does not recognize M.2 NMVe SSD types
Samsung EVO 960 does not show

computerchemist, 2022-11-26:
Confirmed same problem also on ASUS TUF Z590 PLUS with Kingston 2TB M.2 as RAID, shows up fine in latest Crystal Disk Info.

Arigatoss, 2022-12-29:
Doesn't show temps for Kingston KC3000 either.

majorhector, 2023-02-01:
Same on WD Black SN850X. It's a pity

xthree, 2024-01-04:
Same with Sabrent SB-RKT4P-4TB. Sad about it.

khcy4yys, 2024-01-12:
Same with Crucial T500

vlab22, 2024-02-21:
No sensor for WD Black SN850X too

whitedavidp, 2024-07-09:
Running version0.96 on both machines...

On my ASUS PRIME B250M-A based desktop, I cannot see temps for my Samsung SSD 960 and on my HP 8595 (Elitedesk G6 Mini) I cannot see temps for my WDC PC SN730. On both machines, I can see temps for these drives using HD Sentinel. It would be great if I could as well using OHM and its API.

Thanks

itsTashfeen, 2024-11-05:
Still can not see temps on NVMe M.2