openhardwaremonitor/openhardwaremonitor #1541: Not recognising CPU temps

shadders-06, 2022-10-30:
Hi, 
I was wondering why the software won’t pick up on my cpu temps, but will pick up utilisation. I have used OHM on my laptop and it works perfectly, but on my desktop, no temps. Is there any fix? I have an i7-11700K and a GIGABYTE B560M Aorus Pro AX. The weird thing is, other programs like CPUID HWMonitor picks up the temps fine. 

Thanks, Liam

nick0182, 2022-11-01:
@shadders-06 As stated in official documentation of OpenHardwareMonitor `To install just unpack the zip archive and run OpenHardwareMonitor.exe with Administrator rights. Without Administrator rights most hardware sensors are not accessible`
Perhaps you run it without admin rights

shadders-06, 2022-11-01:
Hi, sorry but that hasn't changed anything.

appyface, 2022-11-07:
Hi, I have a Surface Laptop 5 and no temps are shown at all? I was previously using a Surface Pro 3 and all temps displayed for it. Am running OpenHardwareMonitor as administrator (by marking the administrator box on the EXE). I also tried running it by using right-click menu with Run as Administrator. I am also logged in to a local administrator account...

DarthDestroyer, 2022-11-07:
Hey Guys... 

The easiest way to find the Problem is the Report you can generate with File -> Report *Something*
Dont send it, save it.

Open it with an text Editor and you will find the source of evil!

You can even upload it here, that someone could fix the Problem (⁠^⁠^⁠)

appyface, 2022-11-07:
Hi @DarthDestroyer I saved the report, there is nothing in there about temps at all that I can see. And the only error message is for an AMD dll, this system does not have AMD so I think that's OK. Should I paste the report here in a comment?

DarthDestroyer, 2022-11-07:
upload it as full file ;⁠)

appyface, 2022-11-07:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/9956160/OpenHardwareMonitor.Report.txt)

@DarthDestroyer Thanks. Please let me know if I've overlooked something in there?

DarthDestroyer, 2022-11-08:
@appyface 

Microarchitecture: Unknown
Time Stamp Counter Multiplier: 0

12th gen... Seems not declared, yet

mchmaster84, 2023-03-16:
I had this similar issue. Some temps were showing, namely from my SSDs and GPU but not from my CPU or motherboard. After some digging around, I discovered that Avast antivirus was blocking an open hardware driver. If you disable this feature under Troubleshooting > Block Vulnerable Kernel Drivers, I discovered that now all of my temp settings display correctly again.

appyface, 2023-03-16:
@mchmaster84 I don't run any antivirus (except windows defender, which hasn't been a problem in the past?). But perhaps this can help someone else. Thanks for the info.

XaverStiensmeier, 2023-05-13:
I have the same issue with i5-13600K

wcomartin, 2023-10-23:
Same issue with the i5-13600KF

samuelt005, 2024-01-11:
Same issue with I7-14700K

reganola, 2024-01-19:
Same issue with i5-12600k

lcoguss, 2024-01-23:
Same issue with i5-12400

asuSlash, 2024-02-07:
Opened as administrator and i have same issue with i7-13700k. Also i can't see my Asus AIO, but i don't know if this feature is supported
[image]

asuSlash, 2024-02-07:
Runned as administrator and i have same issue with i7-13700k. 
Also i can't see my Asus AIO, but i don't know if this components are supported
[image]

Davideck123, 2024-03-19:
Same issue with i7-12650H.

[image]

That-Ruben, 2024-04-20:
Same issue here, but I can confirm that using LibreHardwareMonitor works fine!

MarcusCasagrande, 2024-04-20:
Same with 14900K here. Gonna try thi Libre version instead

GmanAtheist, 2024-08-03:
i9 11gen MSI not showing temps or fan speeds. Have sent a report. Can this be solved. Running OHM as admin. Thanks.

marcospgp, 2025-05-03:
same issue with i5 12400F, but core temp (a similar app) worked!

wuthimi-jpg, 2025-09-19:
Also had the problem . Started a couple of months ago. Other similar apps still worked. Then I had the same thing happen on a different machine. But this time I was able to catch it right as it happened. Turned out that Windows defender detects a false positive and archives a certain part of the app. After I removed this back out of quarantine the temperatures and frequencies were working again. So I did the same with the previous machine. I found it was the same issue, with Windows Defender. So again I removed the quarantine and it solved the problem. I hope this helps everyone.

gutierrezhector, 2025-09-19:
Just had the same problem. I was able to correct it by authorizing a driver that was blocked by microsoft defender, thanks for reporting it 👍 

It is even suggested by ms doc [there](https://support.microsoft.com/en-us/windows/microsoft-defender-antivirus-alert-vulnerabledriver-winnt-winring0-eb057830-d77b-41a2-9a34-015a5d203c42)