openhardwaremonitor/openhardwaremonitor #1546: MSI A520M-A PRO w/ AMD Ryzen 5 5600G: "Clocks" issues: shows unreasonable values

ralfbergs, 2022-11-14:
Libre Hardware Monitor version 0.9.1

My hardware:

- MoBo MSI A520M-A PRO
- BIOS: 1.B0 (latest as of today)
- CPU AMD Ryzen 5 5600G with Radeon Graphics
- RAM: 32 GB, 2 x 16 GB DDR4-3200, A-DATA Technology

I'm seeing the following issues on my setup for PC ->CPU -> Clocks:

- readings of "NaN" (not-a-number) MHz (comes from "divide by zero?")
- "infinity symbol" MHz; these will not be "sticky" (i. e. show up at the right-most column as the "Max" value)
- frequencies of 15,xxx MHz (fifteen thousands), which seem unreasonably high

Let me know if/how I can help chasing these issues down.

Many thanks for keeping the original project alive with this new fork.

DarthDestroyer, 2022-11-14:
With Libre, you are wrong here... Sorry

ralfbergs, 2022-11-15:
@DarthDestroyer Thanks for letting me know... I don't know where I was with my mind... Had both repos open, and picked the wrong one... ;-(

I've now created https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/892 in favor of this one, and will close this one right now.

closed 2022-11-15 by ralfbergs