openhardwaremonitor/openhardwaremonitor #1549: does not show CPU temperature

JunSourceX, 2022-12-19:
I can't get information about the processor, namely:
cpuTemp
cpu Power Draw Package
cpu Frequency
shows only - CPUUsage. 
What could be the problem? I started it on behalf of the administrator, it didn't help. 
The processor is Intel I5 12400.
I also want to note that it shows all the information about the GPU.

RealHumanWithHair, 2023-01-10:
It is because from 2years have no support in that awesome program, so this mean, the new hardware is not listed to detect new detectors chips etc.

Sadge

JunSourceX, 2023-01-11:
> Это потому, что в течение 2 лет у этой замечательной программы нет поддержки, а это означает, что новое оборудование не указано в списке для обнаружения новых чипов детекторов и т. Д.
> 
> Садж

Why then, if you download a ready-made HWMonitor application, does it show the temperature of my processor?

PCAssistSoftware, 2023-01-11:
> Why then, if you download a ready-made HWMonitor application, does it show the temperature of my processor?

Because a ready-made HWMonitor application (such as HWiNFO) is continuously developed to support new motherboards and processors.