openhardwaremonitor/openhardwaremonitor #1554: To bad the program is out of date :(

RealHumanWithHair, 2023-01-10:


jmauro, 2023-01-19:
Hello,

For your information, there is some kind of fork updated regularly: [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor)

Kind regards,
JM

yusufuygur1, 2023-01-19:
I sended can u check it ? I waiting your payment then I will send code
after you begin use ..

jmauro, 2023-01-19:
> I sended can u check it ? I waiting your payment then I will send code after you begin use ..
> […](#)

What exactly are talking about ?

yusufuygur1, 2023-01-19:
Sorry I wrong send 😂

RealHumanWithHair, 2023-02-16:
> Hello,
> 
> For your information, there is some kind of fork updated regularly: [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor)
> 
> Kind regards, JM

I C

Still have a problem with my CPU, and AVG detect some kind weird stuff. Sometimes indeed antivirus program detecting something which is not a virus but idk.
[image]

but thx you!

sergiohzph, 2023-04-05:
Another case of abandonment in the world of open source software, it's disgusting. I suggest you use HWInfo or HWMonitor which are up to date (because of course they are proprietary). I advise you not to trust on "open source" software because it always ends up being abandoned by the people who are so fond of it and develop it.

moellerm, 2023-06-07:
Open source software is never a promise to get software that is maintained and developed forever. But it is open, anyone can pick it up and continue development or just learn from it.

closed 2023-06-07 by moellerm