openhardwaremonitor/openhardwaremonitor #1557: vulnerability detected by Avast in kernel from OpenHardwareMonitorLib.sys file

nicolas2k, 2023-02-11:
Since few time, I've received a vulnerability detected by Avast in kernel from OpenHardwareMonitorLib.sys file on the version  openhardwaremonitor-v0.9.6.zip : 
[image]

I must to disable this option but it doesn't correct for security of my computer and I lost speed cpu and temperature.
[image]

[image]

I think, it's better to remove remote function

DarthDestroyer, 2023-02-11:
it has to be fully allowed and unblocked!

it will use some functions, which could be use in a bad matter too...

nicolas2k, 2023-02-11:
I've tried to add exclusion but it blocks always ... It seems I've not solution this way of lib.sys

nicolas2k, 2023-02-12:
a runway maybe

https://eclypsium.com/wp-content/uploads/2019/11/Mother-of-All-Drivers.pdf

https://www.safebreach.com/resources/hp-touchpoint-analytics-dll-search-order-hijacking-potential-abuses-cve-2019-6333/

...

Majikeyric, 2023-02-14:
I just encountered the same problem after latest Avast update. I use Windows 7 x64

inquartata, 2023-02-15:
Same here. Same notification from Avast, unable to unblock, lost CPU data. What would be the best way to resolve this?

BenJul79, 2023-02-18:
Same for me. 
As I read on links given in a previous conversation, a solution could be to activate "virtualization-based protection of code integrity". I haven't done it yet because I don't know exactly what it is.

nicolas2k, 2023-02-19:
Hi!

Is it possible to retrieve old version in archives like https://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.zip or 0.7.1 or 0.6.0 ? 

Because, I really need because of the gaps in the laptop which suffers the performance of Intel i7 Gen10 ...

Thanks in advance

Adgenodux, 2023-02-23:
Same problem too after OHM update from 0.9.5 to 0.9.6 1h ago, now Avast refuses it, even when i downgrade to 0.9.5

I've sent the OHM 0.9.6 archive to Avast for testing...

nicolas2k, 2023-02-23:
Hummm, strange, I've tested on different versions, same problem with see picture below 👍 
[image]

nicolas2k, 2023-02-23:
@fonder : no body have old versions in archive to test ? thk

Adgenodux, 2023-02-23:
> Hummm, strange, I've tested on different versions, same problem with see picture below 👍 [image]

yes, it looks like to be an Avast issue...

nicolas2k, 2023-02-23:
sure ! maybe more secure actually with some verifications drivers in kernel ... (why not !)

Adgenodux, 2023-02-23:
hardware monitoring needs to go deeper in the machine, so of course, it makes the antivirus nervous due to security risks...

Waynarky, 2023-02-25:
I'm having the same issue. Please post a resolution if anyone gets one.

inquartata, 2023-02-25:
I couldn't find a solution so I ended up deactivating the avast feature that disabled ohm. That protection is now off (restart Avast if you cannot find it in the settings. The message tells you what part of the protection it is. Webshield or something...) and got OHM working, but of course disabling protection is not exactly a good idea...nor is it really a solution...

Waynarky, 2023-02-25:
OK thanks. Yeah, I'm not sure I want it turned off completely, but Asus is needed. I wonder why its triggering an alert in the first place.

nicolas2k, 2023-02-25:
> 

With less severe safety, I confirm its operation. 
You (@dev) will have to change your method with the Ohm.sys file or find a restraint and necessary library. I'm not expert

Adgenodux, 2023-02-28:
or wait hopefully Avast fix for false positive...

nicolas2k, 2023-02-28:
> false positive

I don't think of false positive, there is a vulnerability of the method directly from memory, see my remark (perhaps) above or here : https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1557#issuecomment-1427047891

Also I think it's minor alert but it block to collecte some measures !

nex86, 2023-03-08:
We are having this problem as well.
We use Avast and servereye (which uses openhardwaremonitor) at a lot of customers and they're all getting the warning now about the blocked driver.
We have reasons to not completely disable the protection.  isn't there a way to make seperate exceptions for specific drivers?
I applied rules to all exceptions on the policity on the Avast business console but all of them are being ignored.

Hopefully this gets sorted out ASAP.

nicolas2k, 2023-03-14:
Hi!

Is it possible to recompile this OHM (minor version) without recreate the file OpenHardwareMonitorLib.sys each opened to avoid detection from Antivirus, to save time ? Because, I didn't found the source OpenHardwareMonitorLib.sys in this GitHub.

Thanks in advance at founder !

keiserS, 2023-03-15:
Hi,
can someone verify this is legit file and it can be added as an exception file in Avast? (without disabling this security option in Avast).

Regards,
K.

nex86, 2023-03-15:
> Hi, can someone verify this is legit file and it can be added as an exception file in Avast? (without disabling this security option in Avast). I have checked the folder and didn't found that file at all (Avast must have removed it to quarantine)
> 
> Regards, K.

Its a .sys file which I already tried to all kind of exceptions including the full path to the file.
Doesn’t work.

nicolas2k, 2023-03-15:
Me too !

One solution temporary or an alternative is : 
- after boot, just before start OHM, disable this case temporary 
- start OHM application
-  and enable this case, it works until next reboot ...

keiserS, 2023-03-15:
> Me too !
> 
> One solution temporary or an alternative is :
> 
> * after boot, just before start OHM, disable this case temporary
> * start OHM application
> * and enable this case, it works until next reboot ...

seems then like a bug report for Avast.
Tnx guys for reply.

K.

Adgenodux, 2023-03-18:
[image]

Tested today CPUID HWMonitor V1.50 and works perfectly without warning nor blocking...

So there is a way...

nex86, 2023-03-18:
> [image]
> 
> Tested today CPUID HWMonitor V1.50 and works perfectly without warning nor blocking...
> 
> So there is a way...

I don't think that HWMonitor uses the same driver/library...

DarthDestroyer, 2023-03-18:
It doesnt... And why should you try to compare Open-Source with Closed-Source and a Company behind it?
Makes no sense!
Licensing and Signing costs a lot... If you are willing to pay it, feel free! But you have to understand, its not even close to be compareable...

Set an Exception or get rid of that crappy antivir...
Sorry to say that this harsh, but there is no other way!

nex86, 2023-03-18:
> Set an Exception or get rid of that crappy antivir... Sorry to say that this harsh, but there is no other way!
Exception can't be set for individual drivers, the protection can only be disabled entirely so its everything or nothing.
And over 100 customers are using the antivir with a running license, its not only a matter of effort to install a new one for everyone but also a matter of cost.

DarthDestroyer, 2023-03-18:
You didnt got the point!

Example, Windows Defender... You can set an Exception for that sys file and use it!

Second Point you missed, you can sign and license drivers and tools... So it looks like legit applications, that costs money!

It doesnt mean any license for that antivir...

PCAssistSoftware, 2023-03-18:
Has anyone tried submitting the file to AVG/Avast to analyze to see if they will mark it as safe?

https://www.avg.com/en-gb/false-positive-file-form#pc

https://www.avast.com/en-gb/false-positive-file-form.php#pc

If the driver genuinally doesn't cause damage and the detection is a false detection then letting them analyze it may allow them to determine it is actually okay and solve the problem.

Just a thought....

nex86, 2023-03-18:
> Example, Windows Defender... You can set an Exception for that sys file and use it!

we're not talking about Windows Defender, we're talking about the self-defense module of Avast and AVG that does NOT provide exceptions for individual kernel drivers.

nex86, 2023-03-18:
> Has anyone tried submitting the file to AVG/Avast to analyze to see if they will mark it as safe?
>
> https://www.avg.com/en-gb/false-positive-file-form#pc [...]

yes already have, no response yet.

PCAssistSoftware, 2023-03-18:
> yes already have, no response yet.

Strange not to get a reply - usually AVG reply quite quickly to requests in my experience - then again I am an AVG reseller so perhaps that is why I get responses.  If you don't get a response let me know and I will try submitting it too

tjsepka, 2023-03-19:
Where did you find the "openhardwaremonitorlib.sys" file that the AV software is flagging?  It's not where it says it is.  I see "openhardwaremonitorlib.dll" in the directory.

PCAssistSoftware, 2023-03-19:
> Where did you find the "openhardwaremonitorlib.sys" file that the AV software is flagging? It's not where it says it is. I see "openhardwaremonitorlib.dll" in the directory.

It doesn't get created if the AV flags and blocks it - turn off your AV blocking and you will see it

[image]

PCAssistSoftware, 2023-03-20:
Details from Avast on the vulnerability - https://www.cvedetails.com/cve/CVE-2020-14979/

tjsepka, 2023-03-21:
It appears that the file is created when the program is run and AVG promptly wipes it.  I guess we'll have to wait for a fix.  I was running version 0.8.0 because it is the only version I found that supports graphing.  While it does run, sesnors that I monitor are no longer shown while others are.

[image]

nicolas2k, 2023-03-22:
don't make a confuse between false virus and vulnerability. In this case, it's not virus detected but there is vulnerability under this method of OHMlib.sys like explanation see (on Feb 12) :
https://eclypsium.com/wp-content/uploads/2019/11/Mother-of-All-Drivers.pdf

https://www.safebreach.com/resources/hp-touchpoint-analytics-dll-search-order-hijacking-potential-abuses-cve-2019-6333/

and like said by Darren here : https://www.cvedetails.com/cve/CVE-2020-14979/

So, we cannot add an exception because it doesn't a viral or false virus. 

have a nice day !

tjsepka, 2023-03-22:
I agree with the assessment above.

helonaut, 2023-04-28:
We are using Atera on a LOT of devices for remote hardware monitoring purposes. Does this mean we have to switch RMM solution? A CVE score of >7 is worrisome to say the least...

nex86, 2023-04-28:
> We are using Atera on a LOT of devices for remote hardware monitoring purposes. Does this mean we have to switch RMM solution? A CVE score of >7 is worrisome to say the least...

Atera still works but you can not monitor things like CPU temperature, (maybe CPU usage?), power consumption and more hardware related things.

CallumAlbert, 2023-05-05:
Another RMM having this issue was/is Syncro. I reported the issue to Syncro and they've now informed me the issue with Avast has been rectified. I no longer appear to be getting the vulnerable driver issue on our machines. I'm unsure as to whether they changed hardware monitoring library or whether they were able to work with Avast to remove the false positive. It might be worth others giving it a go now with the latest definitions and seeing how you go.

nex86, 2023-05-10:
We got a response from Avast, saying they won't add an exception for the driver and the developers of the monitoring software (Atera / servereye) and/or the driver have to fix it.
[image]

moellerm, 2023-06-06:
The last time I checked, the main issue (at least for me and open source projects) preventing any progress in this area is getting a new driver signed.

An attempt to fix this issue was made in this commit:
https://github.com/openhardwaremonitor/openhardwaremonitor/commit/0e435cafc61fe84d429e8a9500d666e34e3de67b

This should restrict access to the driver to privileged accounts once the driver is loaded. Ideally this should be done in the entry code in the driver, as noted in the source code: 
https://github.com/openhardwaremonitor/openhardwaremonitor/blob/09c16898a4a84c36644e7c07baebf1efd299691e/Hardware/KernelDriver.cs#L70
But this would require the resulting driver to get signed again.

I'm not sure if there is a practical way to access (and exploit) the driver from an unprivileged account the way it is loaded in the Open Hardware Monitor. Perhaps there is a way to access it right after starting the service, but before the access gets restricted. The correct way would certainly be to use the IoCreateDeviceSecure in the driver entry code instead of the IoCreateDevice used currently:
https://github.com/openhardwaremonitor/openhardwaremonitor/blob/master/External/WinRing0/OpenLibSys.c#L51

In some vulnerability reports it is claimed that the WinRing0 driver allows to read and write arbitrary memory. To my knowledge this is not the case with the version used in the Open Hardware Monitor, as it seems to have been compiled without the option _PHYSICAL_MEMORY_SUPPORT:
https://github.com/openhardwaremonitor/openhardwaremonitor/blob/09c16898a4a84c36644e7c07baebf1efd299691e/External/WinRing0/OpenLibSys.c#L605

Generally, the architecture used in the Open Hardware Monitor for low level access is not ideal. All lower level hardware access should be done in the driver code and expose only a higher level API that fully validates all input. For example, instead of the current API of "read/write IO port XY" it should be "get temperature XY". But this would require frequent updates of the driver code for every change in supported hardware. And as a consequence as well the frequent signing of newly compiled drivers.

namazso, 2023-06-10:
> In some vulnerability reports it is claimed that the WinRing0 driver allows to read and write arbitrary memory. To my knowledge this is not the case with the version used in the Open Hardware Monitor, as it seems to have been compiled without the option _PHYSICAL_MEMORY_SUPPORT

It allows arbitrary MSR write, which can be abused to overwrite IA32_LSTAR to make the kernel run on user stack. From there on with some ROP, one can achieve kernel code execution, at which point it's a full takeover, equivalent with the access achievable with physical memory mapping. So that option is completely useless from a security perspective.

DenisTimofijuk, 2024-03-14:
This issue still persists. 

Edition	Windows 10 Pro
Version	22H2
Installed on	‎2023-‎03-‎09
OS build	19045.4170
Experience	Windows Feature Experience Pack 1000.19054.1000.0

[image]

PatrickSchmidtSE, 2024-03-14:
Yes, unless someone renews the driver certificate ( one time ) via buying a required certificate and make the microsoft hardware driver check ( to get the microsoft certificate), this will persist.

DrOuille9, 2024-03-31:
May be same problem as OpenRGB ?
https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/2227

D1mar0n, 2024-06-14:
The driver version here is [1.2.0.5](https://github.com/alex-dow/winRing0-rs). Unfortunately, after replacing it with this version, I still get the same error.

pw-in-irvine, 2024-12-17:
Same issue in **Norton 360**.
Submitted to Norton for analysis. No response as of 12/16/2024.

[image]

Neustradamus, 2025-01-23:
Dear @openhardwaremonitor team,

I wish you a Happy New Year 2025!

Have you progressed on it?

IanEarnest, 2025-03-11:
In the past 24 hours there has been an issue with running the program - as listed here https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984

Windows is showing a virus warning of "HackTool:Win32/Winring0"

justmaiklas, 2025-03-11:
[image]

RickyCT, 2025-03-11:
Same issue here, Windows Defender has now also recognised this as a virus and stopping it from running 😭

Cybernetic-Fox, 2025-09-05:
Yeah, just got that this morning

NilsDargent, 2025-09-19:
Same here with Windows Defender (Windows 11), and I was extremely worried, so I hope it's nothing. 
I don't use this, I didn't even know it existence before so I don't know what to do.

[image]

josefelipecenaqueleite, 2026-01-05:
May be has any one at here on this chat with free recovery key from avast .thats a question. I need a lot  ? send private msg to me