openhardwaremonitor/openhardwaremonitor #1559: Incompatible with Windows 11 22H2 System Tray

sivan-koren, 2023-03-10:
If OHM is configured with more than one system tray icon, the tray icon has to be unhidden from the Windows Settings panel or dragged out from the up-arrow expansion (if that's enabled in the Windows Settings).

Started with OS Build 22621.1344.

It seems like Windows doesn't recognize that it's the same program. As if OHM is registering as a whole new tray icon each time it starts.

sivan-koren, 2023-03-11:
Settings look like this after a few restarts.

[image]

smeggie42, 2023-03-16:
This happens to me as well.  im worried it will mess something up by having so many options