openhardwaremonitor/openhardwaremonitor #1560: Driver Antivirus exception (openhardwaremonitorlib.sys)

tjsepka, 2023-03-16:
The program has lost some functionality with AVG stating:

[image]

[image]

nex86, 2023-03-17:
known issue !
https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1557

tjsepka, 2023-03-18:
Issue 1557 is for Avast but this issue is for AVG.  They look similar but only the developer can tell us that.  Issue 1557 was reported mid Ffebuary 2023 but mine started about 2 weeks ago.

Note that the file being flagged by both is not in the location flagged nor part of the program package.  The closest file has a .dll suffix while both packages claim it has a .sys suffix.

Your thoughts are welcomed.

nex86, 2023-03-18:
> Issue 1557 is for Avast but this issue is for AVG. They look similar but only the developer can tell us that. Issue 1557 was reported mid Ffebuary 2023 but mine started about 2 weeks ago.
>
> Note that [...]

From what I notice with Atera client software it tries to place the .sys file in one of its monitor package directories in an interval of 5 minutes but is being blocked and removed by Avast. 
and yes, the folder also contains a .dll file but its not being touched by the antivirus.

tjsepka, 2023-03-19:
I've never had a monitor folder and I can't seem to find one in any of the installation program.  The installation file scans without a problem.

tjsepka, 2023-03-21:
It appears that the file is created in the program's directory when the program is run and AVG promptly wipes it.

wydnas, 2023-03-21:
I also have this issue with AVG, it occurred when upgrading to AVG program v 23.2.3273 (build 23.2.7961.777).  Avast and AVG are from the same company now so the two issues may be related to the same change in both prducts.

PCAssistSoftware, 2023-03-21:
Details from both AVG and Avast on the vulnerability - https://www.cvedetails.com/cve/CVE-2020-14979/

nex86, 2023-03-21:
> Details from both AVG and Avast on the vulnerability - https://www.cvedetails.com/cve/CVE-2020-14979/

I only see something about EVGA precision and winring0  ??

tjsepka, 2023-03-22:
If you look into the "openhardwaremonitorlb.sys" file you will see that it was originally named "winring0.sys".

moellerm, 2023-06-07:
I think at its core are the same problems as in issue #1557. Closing as duplicate.

closed 2023-06-07 by moellerm