openhardwaremonitor/openhardwaremonitor #1564: Virustotal flags this as a trojan?

SvenVD, 2023-04-10:
https://www.virustotal.com/gui/file/5e238c36ae5f8a8ab9aa5e6fa3c568967d61953393384c7c8fd6370f8bc86b85 from https://openhardwaremonitor.org/downloads/

https://www.virustotal.com/gui/file/693256cf15f9e73ad30f9be02f1a6d067fa69ee83ac49438975e14aa183ec2c0/detection from github

Is this normal?

Thanks

nukhes, 2023-04-11:
its just a false positive, open hardware monitor is safe, but outdated, instead, use [libre hardware monitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor), its a regularly updated fork.

moellerm, 2023-06-07:
There are some issues related to security discussed in issue #1557. Not sure what virustotal flags here.

closed 2023-06-07 by moellerm