openhardwaremonitor/openhardwaremonitor #1569: Support for Corsair iCUE Commander Pro & XT

defaultsecurity, 2023-04-22:
Support for retrieving temperature and fan speed data from corsair controllers would be awesome.

- [Corsair iCUE Commander PRO](https://www.corsair.com/eu/en/Categories/Products/Accessories-%7C-Parts/iCUE-CONTROLLERS/iCUE-Commander-PRO-Smart-RGB-Lighting-and-Fan-Speed-Controller/p/CL-9011110-WW)
- [Corsair iCUE Commander Core XT](https://www.corsair.com/eu/en/Categories/Products/Accessories-%7C-Parts/CORSAIR-iCUE-COMMANDER-CORE-XT-Smart-RGB-Lighting-and-Fan-Speed-Controller/p/CL-9011112-WW)

mdstricklin, 2023-05-13:
I second this. I'm using the temp sensor ports on that to monitor my loop water temp and would love to be able to plot that.

defaultsecurity, 2024-05-22:
Any outlook regarding this feature?