openhardwaremonitor/openhardwaremonitor #1578: 13th Gen I7 13700K compatibility

mkanet, 2023-06-16:
Is Openhardwaremonitor still being developed?  I can't seem to find any versions that support current Intel CPUs such as the I7 13700k.  

All I'm trying to do is gain access to my CPU cores temperatures via the Windows 11 cmd or REST query via PowerShell.  I can't find a single solution online that does this.

Plagaf, 2023-06-19:
Got the same issue. I have just found another git that continued the work for a while. I have just downloaded and it looks fine.

[https://github.com/LibreHardwareMonitor/LibreHardwareMonitor](url)

Some recommend HWiNFO

mkanet, 2023-06-19:
Coincidentally, I _just_ started using HWiNFO yesterday.  I was happy to see it has a RESTful web server add-on to access sensor data.  It's a much better way to access HWiNFO's sensor data than my original intention (to use OpenHardwareMonitor's WMI classes to access sensor data).  It's an old add-on but still works very well.

BTW:  I found out that [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor) (still being developed) is a fork of OpenHardwareMonitor.  It looks identical; except it supports newer CPUs like mine.

closed 2023-06-19 by mkanet