openhardwaremonitor/openhardwaremonitor #1589: CPU Usage not reporting properly

soulvoid86, 2023-08-26:
As title says, I have a setup where CPU usage is not correctly reporting in OHM. System is a HPE DL360p Gen9 with E5-2643 v3 CPU's. The system only had 1 CPU, and I just recently added a second CPU, additional memory, etc. Ever since doing so, CPU usage is reporting identical for both CPUs in OHM, and the per core usage is completely wrong. 

I set a prime95 load on 1 node while the other CPU was at 5-10% load, yet the totals for both CPUs showed +/-65% when task manager showed only 1 CPU had high load. 30C temperature difference between CPU packages proves that task manager is accurate. Funny enough temps are reporting properly.

I've cleared all files and started with a fresh config, but the same issue occurs. Worth noting, I have a second identical system with dual E5-2620 v3 CPUs, and that has always reported accurately. I also noticed a difference where with the system that is reporting incorrectly, the CPUs have the exact same name in OHM. But on my second system where OHM is working as expected, each CPU has a corresponding "CPU0" or "CPU1" tag as I'd expect.

No idea what's going on here. All drivers are installed on each system, both are running the same OS, Windows Server 2019 that is fully up to date. Both system's are basically identical besides the one with issues has a quadro GPU, and more memory. (32GB compared to 128GB)

Any help is appreciated!

GameAreaCZ, 2023-10-22:
Can cofirm

SiavaRu, 2023-11-03:
The problem is also present for Intel Core i5-12600KF.
I rebuilt this app, added https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1475 but it didn't helps.

AnOldFart, 2023-11-06:
There seems to be a substancial difference between Windows Resource Monitor CPU load and OHM's on my PC too, the CPU is an i5 9400

digital-spinner, 2024-04-10:
Yep, Ryzen 5900X also off, like about 70% less is shown by Open Hardware Monitor.

soulvoid86, 2024-06-22:
I'm moving over to LibreHardwareMonitor since it seems this project is dead. For my system, the accurate reading is labeled as "CPU Core Max", not "CPU Total", just FYI. That seems to be the inaccurate reading OHM is providing.