openhardwaremonitor/openhardwaremonitor #1603: openhardwaremonitor VS librehardwaremonitor

NolonQ, 2024-01-06:
What EXACTLY is the difference between openhardwaremonitor and librehardwaremonitor?
Judging from the GUI it looks 1:1 the same, does it not?

DarthDestroyer, 2024-01-06:
No versus... Its a Fork? 
It is further in development, because @moellerm isnt doing at the Moment!
That isnt a usefull Issue... if at all

NolonQ, 2024-01-07:
I know it's a fork so what? I had a clear and simple straight forward question, what are the differences between the two, what does the fork change from the original that would me make want to use it, no idea what's there to be complained about usefulness.

DarthDestroyer, 2024-01-07:
Awesome, if you know what is a fork and what it is used for... So you know, that it was the same thing and got further developed by someone else...

Dont know why, but in no case a kind of issue - so what?

justin-octo, 2024-01-08:
The issue is there are no updates for 4 years, but nothing saying why or that it's been abandoned or what the status is.
As far as the difference between this and the fork... you could try to create a PR and then see the difference.  Might have to make your own fork of the two projects and compare.

justin-octo, 2024-01-08:
Assuming you are talking about, https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

NolonQ, 2024-01-10:
> Awesome, if you know what is a fork and what it is used for... So you know, that it was the same thing and got further developed by someone else...
> 
> Dont know why, but in no case a kind of issue - so what?

Congrats for describing what a fork is.
I'll repeat what you just said so you can understand it, here you go "open hw monitor is further developed by someone else".
Yeah, no way? That as vague and unspecific as you can possibly be, if you don't want to help then don't.

NolonQ, 2024-01-10:
Just in case it wasn't obvious enough, i asked a question. Yes a question, not an issue. Drama time?

DarthDestroyer, 2024-01-10:
cute, little guy...

if i have to describe what a fork is, you cant read the timeline, hm? Why should i or anybody explain you code, of you are not willing to understand the smallest fact? 

okay... next round: https://github.com/openhardwaremonitor/openhardwaremonitor/network

read, understand and if you are not able to, google the word Issue next...

Kind regards

NolonQ, 2024-01-13:
I did see the timeline, and? This only means it's outdated which doesn't anser my question.

Never met anyone this cringe on github, and the fact that you can't correctly phrase search the internet but instead are brainwashed by google tell's everything, i'm out

closed 2024-01-13 by NolonQ

DarthDestroyer, 2024-01-13:
lol... Never found anyone such learning, reading and understanding resistent before...

you asked: this vs a fork

You dont want to understand what a fork is and after that you are to incompetent to read a simple graph...

After all, your answer is answered in any way by these two simple words and references!

So, instead of provoking... How are you else? 

Dont see any more brainwashing as by yourself, honey...
Why not use it? 

Kind regards, buddy