openhardwaremonitor/openhardwaremonitor #1611: On Windows 11 24H2 26100.560 didn't detect CPU clocks - just  infinity symbol.

UnclWish, 2024-05-18:


UnclWish, 2024-05-21:
Found a cause of this bug. Windows after update for some reason resets compatibility flag to run as administrator from Openhardwaremonitor.exe
Setting this flag again and running as admin fixes the issue.

UnclWish, 2024-05-22:
Next Windows 11 24H2 update to build 26100.712 again broke bus and cores clocks reading... Running as admin didn't helps... What wrong with this 24H2?..

Taiizor, 2024-05-22:
This project is no longer being developed. I recommend using this instead: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

UnclWish, 2024-05-22:
> This project is no longer being developed. I recommend using this instead: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

I know it. But it's no reason for me to use it. Just some more sensors. And this bug is on it too. Something wrong with .sys file created every launch of program...

Taiizor, 2024-05-22:
> I know it. But it's no reason for me to use it. Just some more sensors. And this bug is on it too. Something wrong with .sys file created every launch of program...

In that case, the root cause of the issue is WinRing0.

UnclWish, 2024-05-22:
Yes, winring0 is the cause. After several launches of Libre and Open HM clocks are read. Can't understand what is going on. But it's somehow linked with 24H2. On 23H2 all works fine.

Taiizor, 2024-05-22:
Actually, I am using 24H2 (26120.670), and the application works without any issues.

[image]

UnclWish, 2024-05-22:
Your build is on other channel. 26120 is insider dev build. 26100 is stable release. I didn't know what's wrong...