openhardwaremonitor/openhardwaremonitor #1619: Open Hardware Remote Monitor appears blank. (no information)

babyyak3002, 2024-07-31:
I've been using Openhardware Monitor for a while now.  Recently I started using the remote web server to monitor temps from my iphone that connects to my network via vpn.  everything was working fine for the first day or two after turning on the remote web server.  

However, now the remote web server just appears blank... there is no info at all displayed.  I've tried deleting and re installing the remote web server.  updating to the latest version.  Start and stop the service.  Nothing has resolved.

anything else I should try to resolve this?   

Windows 10, 64bit
[image]

babyyak3002, 2024-08-08:
any ideas? or any alternatives to try out that have a webserver for monitoring temps?

DarthDestroyer, 2024-08-08:
Have you checked the Developer-Tools?
What is logged in the console?
No Web / ad-Blocker blocking the requests?

The Page itself is loaded, so the "webserver" itself seems working

babyyak3002, 2024-08-08:
I do have pie hole running I can check and see if maybe it's blocking the request. And start there good info! Rick DeMartile

DarthDestroyer, 2024-08-09:
I would check the dev-tools first...

Ctrl+Shift+i... Then check the Network Tab and Console

babyyak3002, 2024-08-12:
Everything just started working again on its own the day you emailed me lol. I'll keep an eye on it. Thank you. For the good vibes! Rick DeMartile

babyyak3002, 2024-08-14:
looks like its doing it again out of no where.
i checked the dev tools network tab and i have these two warnings.  not sure what they mean, i checked the html file and added the doctype command but didn't work after a reboot.
any ideas?