openhardwaremonitor/openhardwaremonitor #1620: missing CPU temperatur section

Vahe4, 2024-08-23:
my computer porcessor is Core i9-12900k 
i want to monitoring my cpu temperature but is missing 
[image]

Vahe4, 2024-08-23:
app version is last 0.9.6

LFAB3, 2024-08-26:
Same for my. Motherboard: x670 with Ryzen 9 9900x

samjviana, 2024-09-15:
> my computer porcessor is Core i9-12900k i want to monitoring my cpu temperature but is missing

The problem is that the last supported Intel CPUs are Tiger Lake's and 12th are Alder Lake. I think this will happen for every newer architecture.
You can solve this by manually editing `OpenHardwareMonitorLib\Hardware\CPU\IntelCPU.cs` ... more specifically adding `AlderLake` to the `Microarchitecture` enum and then referencing it while reading the CPU family and model. And you can do this for any other architecture (feel free to reach out if you need help). I would create a PR for this ... but the repo doesn't seem to be active anymore.

<details>
  <summary>Code</summary>
  
  `OpenHardwareMonitorLib\Hardware\CPU\IntelCPU.cs` line ~18
  
  ```csharp
    private enum Microarchitecture {
      Unknown,
      NetBurst,
      Core,
      Atom,
      Nehalem,
      SandyBridge,
      IvyBridge,
      Haswell,
      Broadwell,
      Silvermont,
      Skylake,
      Airmont,
      KabyLake,
      Goldmont,
      GoldmontPlus,
      CannonLake,
      IceLake,
      CometLake,
      Tremont,
      TigerLake,
      AlderLake
    }
  ```
  
   `OpenHardwareMonitorLib\Hardware\CPU\IntelCPU.cs` line ~224
   
   ```csharp
              case 0x8C: // Intel processors (10nm++)
              case 0x8D:
                microarchitecture = Microarchitecture.TigerLake;
                tjMax = GetTjMaxFromMSR();
                break;
              case 0x97:
              case 0x9A: // Intel Core i3, i5, i7, i9 12xxx
                microarchitecture = Microarchitecture.AlderLake;
                tjMax = GetTjMaxFromMSR();
                break;
              default:
                microarchitecture = Microarchitecture.Unknown;
                tjMax = Floats(100);
                break;
   ```
  
</details>

_Disclaimer: This is (of course) related to Intel CPUs ... but the same behavior can be expected for AMD ones_

tim-99, 2024-09-19:
I just have the .exe. Where can I find the IntelCPU.cs?

Pinkisagit, 2025-02-24:
Thanks @samjviana your suggestion helped me. I've expanded your code to include RaptorLake:
```
  TigerLake,
  AlderLake, // 12th
  RaptorLake, // 13/14th
}
```

Not sure it 0xB6 belong here but I know 0xB7 is right:
```
case 0xB6:
case 0xB7: // Intel Core i3, i5, i7, i9 13xxx and 14xxx (https://en.wikichip.org/wiki/intel/cpuid)
  microarchitecture = Microarchitecture.RaptorLake;
  tjMax = GetTjMaxFromMSR();
  break;
```

There are also two more case statements that need to be updated:
```
case Microarchitecture.Tremont:
case Microarchitecture.TigerLake:
case Microarchitecture.AlderLake:
case Microarchitecture.RaptorLake: {
    uint eax, edx;
    if (Ring0.Rdmsr(MSR_PLATFORM_INFO, out eax, out edx)) {
      timeStampCounterMultiplier = (eax >> 8) & 0xff;
    }
  } break;
```

```
case Microarchitecture.Tremont:
case Microarchitecture.TigerLake:

[OpenHardwareMonitorLib.zip](https://github.com/user-attachments/files/18935463/OpenHardwareMonitorLib.zip)

case Microarchitecture.AlderLake:
case Microarchitecture.RaptorLake: {
    uint multiplier = (eax >> 8) & 0xff;
    coreClocks[i].Value = (float)(multiplier * newBusClock);
  } break;
```

and if statement:
```
    microarchitecture == Microarchitecture.Tremont ||
    microarchitecture == Microarchitecture.TigerLake ||
    microarchitecture == Microarchitecture.AlderLake ||
    microarchitecture == Microarchitecture.RaptorLake) 
{
```

@tim-99, I have compiled the updated code to a  .dll file attached (it has version 0.9.7). You can just replace your existing dll and hopefully it'll work.