openhardwaremonitor/openhardwaremonitor #1633: Virus Detected Issue and wiped from pc essentially

akestrel1987, 2025-03-11:
[image]

Got this warning when I booted my computer today and then was hit with a Virus Detected when trying to reopen it from windows defender. I have had this installed for quite some time and have not seen any updates. Is this project still being worked on and developed or has it been given up on?

[image]

Can't redownload as it is flagged as a virus when it is downloaded and gets auto deleted.

[image]

PCAssistSoftware, 2025-03-11:
Covered in various other issue such as https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1557 - the driver used by OHM has been flagged as a virus for a long time by various other a/v's, obviously not the same for defender.  Until someone can fix this or get a signed version of the driver it is going to remain an issue

akestrel1987, 2025-03-11:
[image]
This is what it is being flagged as by Defender.
I added it to the ignored after turning off the protection redownloading it and then turning it back on for it to flag it and then ignored it. Only workaround I could think of for the time being.

fancyTyphoonKitty, 2025-03-12:
I ran into the same behavior on the safe version tonight. OpenHardwareMonitor v0.9.6 running on Windows 11.

HTD, 2025-03-12:
OK, so it's like false positive, but any clue how to just make Windows Defender to back off from this program? I disabled it to redownload the zip, I run the program, but I have a hunch it will try again to remove the program.

akestrel1987, 2025-03-14:
That is essentially how I did it as well... I disabled RTP redownloaded it put it were I wanted and then re-enabled RTP and when it flagged it I choose to ignore it after it tried to quarantine it.