openhardwaremonitor/openhardwaremonitor #1650: Windows 10 Virus Detection

TomBrandt, 2025-09-05:
Window 10 Defender Virus detection openhardwaremonitor-v0.9.6 OpenHardwareMonitorLib.sys as Trojan:Win32/Vigorf.A Also for Dll file. Downloading App Also Detected Virus.

cubanBword, 2025-09-05:
Just got home today and this was on my computer.  First time this has happened in years of running. can I simply exclude it from windows defender?

[image]

skallx, 2025-09-06:
I got the same thing. Using Win11.

aoakley, 2025-09-06:
I'm also seeing Windows Defender detect and quarantine this.

[image]

neooffs, 2025-09-07:
Hello everyone. In my computer Windows simply quarantined the offending driver (Winring0), not letting me have readings on CPU and Motherboard data.

The driver would let other programs access the system through it, making it not secure in case you download and execute something malicious.

You can whitelist OHM if you are not afraid of executing anything bad.

Here's move info:
https://support.microsoft.com/en-us/windows/microsoft-defender-antivirus-alert-vulnerabledriver-winnt-winring0-eb057830-d77b-41a2-9a34-015a5d203c42
https://www.reddit.com/r/FanControl/comments/1j93doq/why_does_defender_hate_fan_control_an_explanation/

rackemup420, 2025-09-17:
CVE-2020-14979 is the problem, winring0.G has vulnerabilities