openhardwaremonitor/openhardwaremonitor #1653: vulnerableDriver:WinNT/Winring0.G (CVE-2020-14979)

rackemup420, 2025-09-17:
[image]

emmettmikita, 2025-09-18:
[image]

avengerx, 2025-09-19:
Either uninstall OpenHardwareMontior or add an exclusion to the affected file if you really want to use it. The project is no longer being maintained. Winring0 is not really a virus but a known vulnerability, a low-level generic driver written as kind of a sample code that people used as stable library in several projects until its vulnerabilities were exposed. It was never meant to be used in production, if I remember well its story.

It seems the [LibreHardwareMonitor](/librehardwaremonitor/librehardwaremonitor) fork of this project recently implemented a fix (LibreHardwareMonitor/LibreHardwareMonitor#1857) by replacing Winring0 to a PawnIO library. Yet, seems the new implementation still lacks support for several hardware it previously supported.