openhardwaremonitor/openhardwaremonitor #1658: Flagged as virus since a week (started on 22-9-2025)

EddyTheBakker, 2025-09-27:
Unable to run, open, or download - flagged as virus both by Windows Defender (on gaming pc) and McAfee (on server pc)
No specifics given, damn w11 system keeps me in the dark as why or what.

EddyTheBakker, 2025-09-27:
Oh I forgot: Just a few days before the software was flagged as containing a virus, I noticed that the cpu readings were gone, other than loads. 

Asrock X670E Taichi (AM5)
R9 7950X 16-Core
ROG Strix Asus 4090
2x 32Gb DDR5-4800 (64Gb)
Windows 11 Pro, V.24H2, build 26100.6584, FEP 1000.26100.234.0

EddyTheBakker, 2025-09-28:
Update - the file OpenHardwareMonitorLib.dll has the virus (at least, Windows seems to think so) Trojan:Win32/Vigorf.A

Is this a false positive?

mlgmaster666, 2025-09-29:
> Is this a false positive?

Not really, but not a virus either, just a vulnerability, OHM relies on a driver called WinRing0, this driver is very dated and vulnerable to attack, so WinDefender has started to flag anything that uses it.

jschumann56, 2025-09-30:
So mlgmaster666 are we supposed to ignore the issue or wait till it gets fixed to use it?  I am trained (Navy Nuclear) to not ignore warnings.  If you do, you start to ignore them all.  This is a very bad trait.

Can you please do something to fix it?  I have removed OHM from my computers until there is a resolution.

Thanks,
John

defendable-hanspetter, 2025-10-02:
Working as MSSP, can confirm this is an issue. A lot of our customers are experiencing this.
After some analysis I believe this is because the driver is vulnerable to exploitation. Not that OpenHardwareMonitor itself is malicious. By installing this software, with this driver, malicious code can then later come in and escalate their capabilities without going through the correct and secured processes in Windows.

See also
- https://medium.com/@matterpreter/cve-2020-14979-local-privilege-escalation-in-evga-precisionx1-cf63c6b95896 
- https://nvd.nist.gov/vuln/detail/CVE-2020-14979

So this has been an issue for at least 5 years now. If this went under the radar for this project it might be a sign that there should be better control with the supply chain this project is based on.

h0sv1r, 2025-10-20:
inpoutx64.sys... Where did you get this one? I've downloaded Libre ver. and I didn't have it.

dagudoj, 2025-10-23:
Same here. Other programs like Fan Control have changed to a different library.

aneibling, 2025-12-14:
To add detail, [here's ](https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=VulnerableDriver:WinNT/Winring0.A)the vulnerable driver Windows Defender identified for me.